JChristensen/Timezone #5: Support for systems without EEPROM

roadfun, 2016-03-28:
I am working some some ESP8266 hardware that doesn't support the Arduino EEPROM library. But I wanted to support DST. It looks like this library hasn't been updated in a long time. Is it still "alive"?

I'm asking because I modified it so it can be used on the ESP8266, conditionally compiling the code requiring EEPROM. Not sure if it is worth submitting since I don't know if it would ever go into an updated library.

JChristensen, 2016-03-28:
Hi, thanks for asking. It's still alive in that I'm using it but I'm a ways behind (like a year or two) as far as being current with the Arduino IDE, etc. I won't have bandwidth any time soon to make any updates. Still I wouldn't mind seeing your changes, so go ahead with a PR if you like but it'll probably just hang there for some time.

I'm not sure why the EEPROM library is an issue though. Assuming the constructor that reads the rules from EEPROM (i.e. the second one) is not called, then I wouldn't think any EEPROM code would get compiled.

roadfun, 2016-03-28:
Thanks for such a fast response. The problem is the library won't compile on some of the newer boards which lack EEPROM among other features. I.e. telling the IDE to compile for a ESP8266 changes what works/doesn't work at compile time. If you haven't used the IDE in a long time there is now a whole Board Manager that even pulls updates on board capabilities from the Internet.

JChristensen, 2016-03-28:
Right, I'm not up on the board manager or the library manager. Thanks and best of luck.

roadfun, 2016-03-28:
I’m not really up on Github but I did send in a pull request, hopefully correctly :) Might help someone else out eventually.

JChristensen, 2016-03-28:
Hmmm, not seeing the PR. But I had a look at your repo, those changes are fine, straightforward enough.  Maybe I missed the PR before, I'll reopen this issue and check.

JChristensen, 2016-03-28:
Still not seeing a PR.  Give it another shot and if it still doesn't make it, it'll only take a few minutes to recreate your changes.  Thanks!

roadfun, 2016-03-28:
OK, I tried again…

JChristensen, 2016-03-31:
Looks like that did the trick.  Thanks again!

JChristensen, 2018-01-03:
Fixed by https://github.com/JChristensen/Timezone/commit/510ae2f6b6fd64a2fc83ed5cee1c31c452daa7a1

closed 2018-01-03 by JChristensen