JChristensen/Timezone #10: (DST) changes do not necessarily occur on the same date every year.

ali32b, 2016-11-05:
Daylight Saving Time (DST) changes do not necessarily occur on the same date every year.

JChristensen, 2016-11-05:
Yes, I am very much aware of this and the library is so coded. If there is an actual issue, please submit a [SSCCE](http://www.sscce.org/).

closed 2016-11-05 by JChristensen

ali32b, 2016-11-06:
Let me give you and example.This library works for countries that their DST happens exactly on same weekday(ie monday) every year.For countries like Iran(Tehran) DST starts on "monday" 2016 while on 2017 starts on "wednesday".So we can not create TimeChangeRule  to work every year.I hope I've explained that well.
[https://www.timeanddate.com/time/zone/iran/tehran](url)

JChristensen, 2016-11-06:
Thanks for the clarification, I was not familiar with IRDT nor did I understand what country you were referring to.
Since the change dates for IRDT do not follow a rule, the only advice I could offer would be to code a rule for each year and then reference the proper rule based on the current year. Alternately, use a single rule and code a table of change dates which is used to modify the rule on the fly.