JChristensen/Timezone #17: Israel Summer Time

M1ndHunteR, 2017-07-19:
Hello,

I'm trying to use the library for switching ON and OFF summer time in Israel.
Israel uses rather different summer time dates:

Beginning on the Friday before the last Sunday in March (at 2:00 local time, or 0:00 UTC, clocks are moved forward by one hour)
Ending on the first Sunday after October 1 (at 2:00 IDT or 23:00 UTC Saturday, clocks are moved back by one hour).
(https://en.wikipedia.org/wiki/Israel_Summer_Time)

The ending is not the issue, but how can I define the beginning?

Many thanks.

nsayer, 2017-07-19:
This may not help, but if you have code to get the date of the first Sunday...

d = first Sunday in March
while(d + 7 <= 31) d += 7;
d -= 2;

d is now the date of the Friday before the last Sunday.

datacute, 2017-07-20:
The library would have to be changed to support -46 as the hour of the day that summer time starts.

M1ndHunteR, 2017-07-24:
Thanks a lot, **nsayer** and **datacute**!!!!

JChristensen, 2018-01-03:
Outside the scope of this library.

closed 2018-01-03 by JChristensen

doanerock, 2018-01-03:
This is a shame that Timezone can not handle all time zones.

JChristensen, 2018-01-03:
Indeed. Someone else will have to maintain a [tz database](https://en.wikipedia.org/wiki/Tz_database) for the Arduino world. My position [here](https://gist.github.com/JChristensen/f390dea4b1f713b245918500ac45cc31).