JChristensen/Timezone #21: Possibility of adding a specific day of the month rule

azevedo-manuel, 2017-08-15:
Would it be possible to have rules that instead of DOW, change on a specific day/month pair?
I heard about this requirement, but according to Wikipedia, Sweden used it in the past, but only Iran uses this kind of DST today.

azevedo-manuel, 2017-09-10:
I added a dirty hack to support fixed dates
If adding __tzFixedRuleOffset__+__dayOfMonth__ instead of __dweek__ (which accepts values from 1 to 7), it will identify the rule as a fixed date rule.

It's not a clean implementation, but won't break existing **libTimeZone** implementations

https://github.com/azevedo-manuel/Timezone/commit/92b493f236c8cae8c98358e60372e727691d92eb
So far seems to be working correctly!

JChristensen, 2018-01-03:
Outside the scope of this library.

closed 2018-01-03 by JChristensen

doanerock, 2018-01-03:
This is a shame that Timezone can not handle all time zones even thought people are willing to pitch in to help add the functionality.

JChristensen, 2018-01-04:
Yes indeed. Maybe someone else will step up to create and maintain a [tz database](https://en.wikipedia.org/wiki/Tz_database) for the Arduino world. I am [unable](https://gist.github.com/JChristensen/f390dea4b1f713b245918500ac45cc31).