JChristensen/Timezone #24: No work with ESP8266 and Version 2.4.0.rc2

bobybobybob, 2017-10-30:
Hello,
no working with 
2.4.0.rc2 core
2.4.0.rc1 core

WorldClock Demo say Ardiuno IDE 1.8.5

exit status 1
'dowSunday' was not declared in this scope

2.3.0 core  OK   and yes, #include <TimeLib.h>

zoomx, 2017-10-31:
I made a test just now
2.3.0 core and wordclock example. I got the same error using time.h and got a full compilations using TimeLib.h
I guess that in 2.4.0 core there is another TimeLib.h. Try enabling verbose output during compiling in Preferences.

milkpirate, 2017-11-18:
Same problem here. Any solution so far? Verbose output:

```
"C:\Users\Lenny\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Lenny\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/include" "-IC:\Users\Lenny\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/lwip/include" "-IC:\Users\Lenny\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Lenny\AppData\Local\Temp\arduino_build_502467/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10800 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Users\Lenny\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266" "-IC:\Users\Lenny\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\variants\nodemcu" "-IC:\Users\Lenny\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266WiFi\src" "-IC:\Users\Lenny\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ArduinoOTA" "-IC:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\ArduinoJson\src" "-IC:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Streaming" "-IC:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone" "-IC:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Time" "-IC:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\NTPClient" "-IC:\Users\Lenny\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\Ticker" "-IC:\Users\Lenny\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\SPI" "-IC:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\async-mqtt-client-master\src" "-IC:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\ESPAsyncTCP-master\src" "-IC:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\int_defines" "-IC:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\wifi_credentials_P" "-IC:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\ota_credentials_P" "-IC:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\mqtt_settings" "-IC:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\djb_hash" "-IC:\Users\Lenny\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266mDNS" "C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp" -o "C:\Users\Lenny\AppData\Local\Temp\arduino_build_502467\libraries\Timezone\Timezone.cpp.o"
C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp: In member function 'time_t Timezone::toLocal(time_t)':

C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp:45:17: error: 'year' was not declared in this scope

     if (year(utc) != year(_dstUTC)) calcTimeChanges(year(utc));

                 ^

C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp:48:36: error: 'SECS_PER_MIN' was not declared in this scope

         return utc + _dst.offset * SECS_PER_MIN;

                                    ^

C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp:50:36: error: 'SECS_PER_MIN' was not declared in this scope

         return utc + _std.offset * SECS_PER_MIN;

                                    ^

C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp: In member function 'time_t Timezone::toLocal(time_t, TimeChangeRule**)':

C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp:62:17: error: 'year' was not declared in this scope

     if (year(utc) != year(_dstUTC)) calcTimeChanges(year(utc));

                 ^

C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp:66:36: error: 'SECS_PER_MIN' was not declared in this scope

         return utc + _dst.offset * SECS_PER_MIN;

                                    ^

C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp:70:36: error: 'SECS_PER_MIN' was not declared in this scope

         return utc + _std.offset * SECS_PER_MIN;

                                    ^

C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp: In member function 'time_t Timezone::toUTC(time_t)':

C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp:102:19: error: 'year' was not declared in this scope

     if (year(local) != year(_dstLoc)) calcTimeChanges(year(local));

                   ^

C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp:105:38: error: 'SECS_PER_MIN' was not declared in this scope

         return local - _dst.offset * SECS_PER_MIN;

                                      ^

C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp:107:38: error: 'SECS_PER_MIN' was not declared in this scope

         return local - _std.offset * SECS_PER_MIN;

                                      ^

C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp: In member function 'boolean Timezone::utcIsDST(time_t)':

[... cut ...]
C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp:179:43: error: 'weekday' was not declared in this scope

     t += (7 * (w - 1) + (r.dow - weekday(t) + 7) % 7) * SECS_PER_DAY;

                                           ^

C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\Timezone\Timezone.cpp:179:57: error: 'SECS_PER_DAY' was not declared in this scope

     t += (7 * (w - 1) + (r.dow - weekday(t) + 7) % 7) * SECS_PER_DAY;

                                                         ^
```

jackjansen, 2017-11-21:
I had exactly this same problem, the issue is that, when building for esp8266, a different <Time.h> is included.

Changing `Timezone.h` line 19 from
```
#include <Time.h>
```
to 
```
#include <TimeLib.h>
```
fixes it.

sidoh, 2017-12-04:
@willjoha fixed this in #8.  Changing the include worked for me.

closed 2018-01-09 by JChristensen

bobybobybob, 2018-01-09:
2.4.0 of ESP8266 for IDE is out but probem is same... and compile does not work with change time.h to TimeLib.h

commit f928ae74 (2018-01-09): Move source files to /src.
Closes #7. Closes #8. Closes #11.  Closes #14. Closes #20. Closes #24.