JChristensen/Timezone #27: symlink CONTRIBUTING.md breaks checkout under Linux

borisneubert, 2018-01-13:
git clone produces the error:

`error: unable to create symlink CONTRIBUTING.md: Input/Output Error`

with git version 2.14.1 under Linux on a filesystem that supports symlinks.

Please see  [here](https://stackoverflow.com/questions/18411200/git-unable-to-create-symlink-file-name-too-long) or a potential fix

JChristensen, 2018-01-13:
That's interesting because while I did play with a symlink for that file, it was not a symlink at the time of the commit and is not currently on my system.

borisneubert, 2018-01-13:
[image]

The symbol in the GitHub UI suggests that it is a symlink. Above mentioned link makes me think that the issue is hard to detect and resolve.

JChristensen, 2018-01-13:
Yes I did notice that. I removed the file, committed, then added it back and committed again.
Please let me know if that fixes it.

borisneubert, 2018-01-13:
Now it works fine for me. Thank you!

JChristensen, 2018-01-13:
Great, thank you for the catch!
Fixed by 3057e97e1eeb4f00357db7800fa432edd188a0d8

closed 2018-01-13 by JChristensen