JChristensen/Timezone #28: IANA timezone standards

myk3y, 2018-01-19:
Wouldn't it be better to use IANA timezone standards, rather than US ones like EDT, etc?

Such as Americas/Chicago or Asia/Singapore.

That would enable looking up your local time source and getting the API to return to you the timezone you are in.

It also makes it more internationally acceptable - NZDT is all very well if you are in NZ and BST if you are in the UK, but to be truly international, your library should use agreed-upon timezone standard notation.

JChristensen, 2018-01-19:
The Timezone library is aware of neither the IANA names nor of other abbreviations like EDT. It does not contain any database of timezones. It only knows what it is told via the TimeChangeRules, and the timezone abbreviation used is completely up to the user. However, the abbreviation is limited to five characters so the longer IANA names would not fit without changes.
The Timezone library was written for small microcontrollers with very modest resources, hence these design decisions.

closed 2018-01-19 by JChristensen

myk3y, 2018-01-27:
So really it is a time-based trigger, nothing to do with timezones at all.

JChristensen, 2018-01-27:
Characterizing it as "nothing to do with timezones at all" strikes me as a bit odd but I suppose the bottom line is to understand how the library works and whether it's suitable for a particular application.