JChristensen/Timezone #29: Timezone.h works no longer for esp82366 board 2.4.0, but with  board 2.3.0 it is still fine

tofrnr, 2018-03-05:
hello
why  compiles Timezone.h 
https://github.com/JChristensen/Timezone
with esp board 2.3.0 but not for 2.4.0 ?
perhaps which different lib or board revision?
( the Arduino Time lib is installed and included.)

error messages for 2.4.0 (2.3.0. is fine though) :

```
D:\Arduino\libraries\Timezone-master\Timezone.cpp:182:20: error: expected primary-expression before ')' token

     t = makeTime(tm);        //first day of the month, or first day of next month for "Last" rules
                    ^
D:\Arduino\libraries\Timezone-master\Timezone.cpp:182:20: error: 'makeTime' was not declared in this scope

D:\Arduino\libraries\Timezone-master\Timezone.cpp:183:43: error: 'weekday' was not declared in this scope

     t += (7 * (w - 1) + (r.dow - weekday(t) + 7) % 7) * SECS_PER_DAY;

D:\Arduino\libraries\Timezone-master\Timezone.cpp:183:57: error: 'SECS_PER_DAY' was not declared in this scope
```

JChristensen, 2018-03-05:
Are you using this core?
https://github.com/esp8266/Arduino

tofrnr, 2018-03-05:
tbh,  I assume it's that core which you mentioned, I got the core originally from the Arduino (tool) board management; available are e.g.
2.3.0 (ok +)
2.4.0 (fails)
2.4.0-rc1 (fails)
2.4.0-rc2 (fails)

JChristensen, 2018-03-05:
I am not currently working with ESP8266 modules, but I installed the 2.4.0 core, and the Clock example from the Timezone library compiled clean for me. From the error messages, I'd say there may be some corruption either in the Time library (TimeLib.h) and/or with the Timezone library. Looking at the error messages, makeTime(), weekday(), SECS_PER_DAY, etc. are defined in the Time library. Also, the line of code reported as line 183 [is actually line 182](https://github.com/JChristensen/Timezone/blob/master/src/Timezone.cpp#L182) so that makes me wonder too.

tofrnr, 2018-03-05:
ok, I'll report the ssue also at the time github repo.
Nonetheless, stangely the errors are about Timezone-master\Timezone strangely.

But what would you think - why can that happen just for all 2.4.0 while 2.3.0 is fine?

JChristensen, 2018-03-05:
I have no idea as I can't duplicate the issue. I also might not raise any more issues until I verified my installation.

JChristensen, 2018-03-05:
I see no reason to suspect the Time library.

closed 2018-03-05 by JChristensen

per1234, 2018-03-05:
It's a duplicate of https://github.com/JChristensen/Timezone/issues/24 (and all the other duplicate issues). The problem is that @tofrnr is using an outdated version of the Timezone library, which still has the `#include <Time.h>`.

JChristensen, 2018-03-05:
@per1234, thank you.
@tofrnr, try replacing your Time library with this one:
https://github.com/PaulStoffregen/Time
It's also in the Arduino Library Manager, v1.5.0

JChristensen, 2018-03-05:
Oh wait, I misread that. @tofrnr please get the latest version of the Timezone library, 1.1.1.

tofrnr, 2018-03-05:
the time lib https://github.com/PaulStoffregen/Time was already my current version.

Nonetheless, per123 was right about an older Timezone lib, I was using 1.0 (Mar 2012) 
and now updated to 

name=Timezone
version=1.1.1
( also Mar 2012 ac. to timezone.h ?  )

Now I just get some warnings, e.g., 

> WARNUNG: Bibliothek Timezone-master behauptet auf (avr) Architektur(en) ausgeführt werden zu können und ist möglicherweise inkompatibel mit Ihrem derzeitigen Board, welches auf (esp8266) Architektur(en) ausgeführt wird.

 but currently no errors any more, 
I'll watch that more closely for different other projects of mine  and perhaps will report then later.

Thank you very much for your reply so far!

JChristensen, 2018-03-05:
Glad that it's working now. Like I said, I haven't gotten into the ESP modules yet, so my libraries are just tagged for the AVR architecture and hence the warning message. Normally I will only be able to give low priority to issues with other architectures.