JChristensen/Timezone #31: request: update LICENSE.md file to enable git licensing support

bperrybap, 2018-03-15:
If LICENSE.md contains the clean text of the license, git will display the license on the main repository page.
It is a really nice feature of github.
All you would need to do is copy your preferred license text into your LICENSE.md file.
CC 4.0 license, are here: https://creativecommons.org/2014/01/07/plaintext-versions-of-creative-commons-4-0-licenses/

After that, github will show the license and put a link to your license.md file and will also provide a highlight as to the licensing requirements whenever the license file is selected/read on the site.
Yeah, you loose the ability to show a cute icon in the license file but that is probably better handled in the README file anyway.
Here is a dummy repo I created to demonstrate: https://github.com/bperrybap/junk

JChristensen, 2018-03-16:
Thanks, I'd seen that on other repos but hadn't looked into it. I think CC is OK for my hardware projects but I should change to GNU GPLv3 for my code repos.

bperrybap, 2018-03-16:
I'm a big fan of GPLv3 for my own projects.

closed 2018-03-21 by JChristensen

Gamadril, 2018-06-24:
IMHO GPLv3 is a killer for that library. That means that each sketch that uses Timezone library and put to pubic domain must be GPL as well. Commercial projects actually must not use it at all if they don't want to make own code public.
My sketches have the MIT license which is not compatible with GPL, means I cannot use the library...

joeyhagedorn, 2018-11-01:
I'm also quite disappointed with this change. I started using the library (and your DS3232C library) prior to the change to the GPLv3, and in the process of getting ready to document and share my project I discovered this. I prefer to publish my work under the MIT license and It looks like I need to rewrite my stuff now not to use this library before publication.

Are you strongly attached to this license which requires anything linking it to use GPL, or would you consider providing your libraries under the LGPL perhaps?

JChristensen, 2018-11-01:
I don't like that the MIT and LGPL licenses permit use within proprietary software. Perhaps you can consider using GPL.

bperrybap, 2018-11-02:
I really don't think the license changed that dramatically
In fact, prior to its change to GPL the library code was not usable in any commercial product.
CC (the previous license for this library) is actually worse than GPL.
CC is only compatible with itself and hence is not suitable for s/w since
With CC licensed code you can't legally link it with anything other than CC code.
i.e. can't be linked with other GPL, LGPL, BSD, or MIT etc code.... Didn't matter.
This meant that CC licensed s/w could never be used in any commercial product since most  products use some other code that is not licensed CC.
The CC license was updated with CC version 4.0 to allow a one time conversion away from CC to GPL 3.0 so CC 4.0 was effectively the same as GPL v3 as far as a commercial product was concerned.

I've been on both sides of the GPL fence. I started a company and for my company I wanted the most liberal license possible for business reasons. As an open source author I fall on the other side of the fence where I don't want commercial products to profit off my efforts by incorporating into closed source.

JChristensen, 2018-11-02:
@joeyhagedorn, if we're talking about a few gifts for friends and family, then why is what license you choose for your code a big deal? These aren't commercial product, yes?

joeyhagedorn, 2018-11-02:
If someone else runs across my code, finds something they like, and wants to copy/paste it into their project, I’d like them to feel free to do so plain and simple. It’s not a commercial product, I’ve never sold one, and do not intend to sell one.

JChristensen, 2018-11-02:
My goals are the same, plus I want to encourage open source. My understanding is that with GPL pretty much all usage is OK, with the exception of closed-source commercial projects.

JChristensen, 2018-11-02:
The current version of the library is GPL, not CC.

I appreciate the conversation too. My hope was that GPL would not be a major stumbling block for folks that want to use the library. But open source is important to me as well. Heck, I might go with the Beerware license if it stipulated open source.

A couple of my libraries get enough play that I wanted some license, if only for the warranty exclusions. I'm only aware of a couple times my libraries were used in commercial or public projects. I don't know what I don't know, but surely 99% of the use is hobbyists that probably never publish their code.

I wish there was a way you could feel comfortable using the library, at least until that highly unlikely and hypothetical case comes along ;-)

commit 78040750 (2018-03-21): GNU GPL v3.0.
Closes #31.
commit e8e3d723 (2018-03-21): GNU GPL v3.0.
Closes #31.
commit 3713faff (2018-03-21): GNU GPL v3.0.
Closes #31.