JChristensen/Timezone #34: In library.properties, architectures=avr throws warning when compiled on ESP8266

gitpeut, 2018-03-31:
When compiling anything with the Timezone library using board ESP8266 version 2.4.1, I get the following warning:

`WARNING: library Timezone claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (esp8266) architecture(s).`

Ignoring this warning is a fine workaround. However, it is distracting when cleaning up code.

Reason for this warning is this last line in library.properties:
`architectures=avr`
I have had no warnings or issues after changing this line to
`architectures=*`

As Timezone is a not very hardware specific library, I guess the risk of doing so for new or yet unknown
platforms is minimal. Please consider including this change in a future release of the Timezone library.

JChristensen, 2018-03-31:
I work strictly with the AVR architecture, and while I'm aware that this library apparently works on other architectures, I don't advertise it as such since I'm not in a position to try and duplicate issues, etc. OTOH I'm not totally opposed to making some changes and tweaks to accommodate other architectures and I've done so in several instances. But until and unless I start using the library on other architectures, it will stay tagged for AVR only. I take this approach partly out of a desire to provide quality code that works as advertised, and partly to limit the time required to respond to issues, etc. as I often don't have a lot of bandwidth. Sorry for the inconvenience; I hope this makes some small amount of sense.

closed 2018-03-31 by JChristensen

gitpeut, 2018-03-31:
Hi Jack,

I understand. Thanks for your efforts and your quality work.

Regards,
Jose

woutput, 2019-01-02:
Hi @JChristensen 
I understand your reasoning. In addition I think it is quite a nice compliment if multiple people see the use of the library and also want to use it on other CPUs :)
Would it be possible to make this library support other architectures (preferably starting with ESP8266) if other people would help? And if yes: what is the work you would like be done?

per1234, 2019-01-02:
@woutput just make a hard fork of the library. Trying to get any PR merged to this repo for non-AVR is just a waste of your time because it will never get merged.

I already did the work for ESP8266 and ESP32 here: https://github.com/JChristensen/Timezone/pull/44
but note a small change needs to be made to that PR, as described [here](https://github.com/JChristensen/Timezone/pull/44#issuecomment-450595627).

The library actually already works on any architecture. The exception is the writeRules and readRules functions and the `Timezone::Timezone(int address)` constructor, which require EEPROM but are not essential. My work in https://github.com/JChristensen/Timezone/pull/44 will also provide support for any architecture that has a compatible EEPROM library, you just need to update the preprocessor conditionals. I don't think there is an EEPROM library for SAMD or SAM. There is a library for SAMD that makes it easy to write to flash and that could be used instead of the EEPROM library for that architecture. I don't know what the situation is for STM32 and nRF51.

per1234, 2019-01-05:
>the management of the library has failed to meet with your approval.

I never said anything of the sort. I'll I did was state verifiable facts, as you've just confirmed once again