JChristensen/Timezone #35: WARNING: library Timezone claims to run on (avr) architecture(s)

microwiz, 2018-06-25:
'When using the library with ESP8266 (Arduino) following warning is observed:

WARNING: library Timezone claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (esp8266) architecture(s).

The library works fine but it could be improved or extended to architectures other than AVR.

closed 2018-06-25 by JChristensen

witnessmenow, 2018-09-09:
Duplicate of #34 in case anyone comes here and wonders why its closed (like me :) )

profharris, 2025-09-27:
Thank You so much.  I was getting the WARNING: Library Qmi8658c claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
       
Following your advice: You can either ‘ignore’ the Warning, the library works well on ESP32 architectures... —Or— to make the Warning go away...
       Edit the “library.properties” file in your libraries\Qmi8658c
            Change the line: architectures=avr
                         To: architectures=avr,esp8266,esp32

BONUS: When you make the change to the “library.properties” file as
       mentioned above, your other Warnings...
         qmi8658c.resultToString(qmi8658_result), 
         Warning: ISO C++ forbids converting a string constant to 'char*'

--These Warnings also go away! Love it when a good plan comes together! 
--Can someone explain why the Warning: “ISO C++ forbids converting a string constant to 'char*'” 
  stopped when I simply added ‘esp8266,esp32’ to the ‘architectures’ line in “library.properties”?
-mpharris@hotmail.com