JChristensen/Timezone #37: Daylight Saving Time

yknivag, 2018-09-24:
I'm having a bit of an issue working out how to set up the TimeChangeRule.

In the UK DST starts at 02:00 on the last Sunday of March where that Sunday is no later than the 30th of March. (If Sunday is the 31st then the change happens the week before!)  The same rule applies for the end in October.

How can I set this as it isn't always the same Sunday?

JChristensen, 2018-09-24:
@yknivag: Interesting, I was not aware of this! I'm sure it can be accomplished with some programming but I'm having trouble finding a reference for the UK DST rules that show this. Can you provide a reference? I found https://www.gov.uk/when-do-the-clocks-change which shows DST starting on 31Mar2019.

JChristensen, 2018-09-24:
The [tz database](https://www.iana.org/time-zones) also seems unaware of the rule excluding 31Mar and 31Oct:
```
$ zdump -v -c2018,2030 Europe/London | grep 31
Europe/London  Sun Mar 31 00:59:59 2019 UT = Sun Mar 31 00:59:59 2019 GMT isdst=0 gmtoff=0
Europe/London  Sun Mar 31 01:00:00 2019 UT = Sun Mar 31 02:00:00 2019 BST isdst=1 gmtoff=3600
Europe/London  Sun Oct 31 00:59:59 2021 UT = Sun Oct 31 01:59:59 2021 BST isdst=1 gmtoff=3600
Europe/London  Sun Oct 31 01:00:00 2021 UT = Sun Oct 31 01:00:00 2021 GMT isdst=0 gmtoff=0
Europe/London  Sun Mar 31 00:59:59 2024 UT = Sun Mar 31 00:59:59 2024 GMT isdst=0 gmtoff=0
Europe/London  Sun Mar 31 01:00:00 2024 UT = Sun Mar 31 02:00:00 2024 BST isdst=1 gmtoff=3600
Europe/London  Sun Oct 31 00:59:59 2027 UT = Sun Oct 31 01:59:59 2027 BST isdst=1 gmtoff=3600
Europe/London  Sun Oct 31 01:00:00 2027 UT = Sun Oct 31 01:00:00 2027 GMT isdst=0 gmtoff=0
$ 
```

JChristensen, 2018-10-02:
Nothing heard; issue closed.

closed 2018-10-02 by JChristensen