JChristensen/Timezone #41: NZDT time calculation error

stinkpot, 2018-11-11:
Hi JChristensen:
During using the timezone in UNO system, it is found the local time of NZDT wrong. The timezone rule is set like this:
TimeChangeRule GDS_general::unoDST = {"NZDT", Last, Sun, Sep, 2, +780};    //Daylight time = UTC + 13 hours
TimeChangeRule GDS_general::unoSTD = {"NZST", First, Sun, Apr, 2, +720};   //Standard time = UTC + 12 hours
Timezone GDS_general::unoTZ=Timezone(unoDST, unoSTD);

Now, these days, the local time is still 12 hours + UTC.
And further debug show:
1, timezone use North local,
2, the last week, maybe cause some problem in function time_t Timezone::toTime_t(TimeChangeRule r, int yr)
these lines:
    if (r.week == 0)
	{
		t -= 7 * SECS_PER_DAY;
		Serial.print(F("last week: "));
		Serial.println(t, DEC);
	}

also it was found with this bug (maybe), the month/day of the local time will going to crazy value, at this stage I convert to DateTime structure( using RTC lib). But still debug this timezone calculation wrong issue.

Can you help to have a look? Not very understand the 'lask week' logic in this function.
Thanks.
If make fix will update to you.
Thanks again.

JChristensen, 2018-11-11:
Looks like it's working OK for me. Please run the following sketch and compare your output to mine:
```c++
#include <Timezone.h>   // https://github.com/JChristensen/Timezone

// New Zealand Time Zone
TimeChangeRule myDST = {"NZDT", Last, Sun, Sep, 2, 780};    // UTC + 13 hours
TimeChangeRule mySTD = {"NZST", First, Sun, Apr, 2, 720};   // UTC + 12 hours
Timezone myTZ(myDST, mySTD);

TimeChangeRule *tcr;        // pointer to the time change rule, use to get TZ abbrev

void setup()
{
    Serial.begin(115200);

    // Simulate Apr-2018 time change
    setTime(01, 57, 00, 1, 4, 2018);
    time_t utc = now() - 13UL * 3600;
    for (int i=0; i<6; i++)
    {
        time_t local = myTZ.toLocal(utc, &tcr);
        printDateTime(utc, "UTC");
        printDateTime(local, tcr -> abbrev);
        Serial.println();
        utc += 60;
    }
    Serial.println("--------------------------------");
    
    // Simulate Sep-2018 time change
    setTime(01, 57, 00, 30, 9, 2018);
    utc = now() - 12UL * 3600;
    for (int i=0; i<6; i++)
    {
        time_t local = myTZ.toLocal(utc, &tcr);
        printDateTime(utc, "UTC");
        printDateTime(local, tcr -> abbrev);
        Serial.println();
        utc += 60;
    }
    Serial.println("--------------------------------");

    // Simulate Apr-2019 time change
    setTime(01, 57, 00, 7, 4, 2019);
    utc = now() - 13UL * 3600;
    for (int i=0; i<6; i++)
    {
        time_t local = myTZ.toLocal(utc, &tcr);
        printDateTime(utc, "UTC");
        printDateTime(local, tcr -> abbrev);
        Serial.println();
        utc += 60;
    }
    Serial.println("--------------------------------");
    
    // Simulate Sep-2019 time change
    setTime(01, 57, 00, 29, 9, 2019);
    utc = now() - 12UL * 3600;
    for (int i=0; i<6; i++)
    {
        time_t local = myTZ.toLocal(utc, &tcr);
        printDateTime(utc, "UTC");
        printDateTime(local, tcr -> abbrev);
        Serial.println();
        utc += 60;
    }
    Serial.println("--------------------------------");}

void loop()
{
}

// format and print a time_t value, with a time zone appended.
void printDateTime(time_t t, const char *tz)
{
    char buf[32];
    char m[4];    // temporary storage for month string (DateStrings.cpp uses shared buffer)
    strcpy(m, monthShortStr(month(t)));
    sprintf(buf, "%.2d:%.2d:%.2d %s %.2d %s %d %s",
        hour(t), minute(t), second(t), dayShortStr(weekday(t)), day(t), m, year(t), tz);
    Serial.println(buf);
}
```
My output:
```
12:57:00 Sat 31 Mar 2018 UTC
01:57:00 Sun 01 Apr 2018 NZDT

12:58:00 Sat 31 Mar 2018 UTC
01:58:00 Sun 01 Apr 2018 NZDT

12:59:00 Sat 31 Mar 2018 UTC
01:59:00 Sun 01 Apr 2018 NZDT

13:00:00 Sat 31 Mar 2018 UTC
01:00:00 Sun 01 Apr 2018 NZST

13:01:00 Sat 31 Mar 2018 UTC
01:01:00 Sun 01 Apr 2018 NZST

13:02:00 Sat 31 Mar 2018 UTC
01:02:00 Sun 01 Apr 2018 NZST

--------------------------------
13:57:00 Sat 29 Sep 2018 UTC
01:57:00 Sun 30 Sep 2018 NZST

13:58:00 Sat 29 Sep 2018 UTC
01:58:00 Sun 30 Sep 2018 NZST

13:59:00 Sat 29 Sep 2018 UTC
01:59:00 Sun 30 Sep 2018 NZST

14:00:00 Sat 29 Sep 2018 UTC
03:00:00 Sun 30 Sep 2018 NZDT

14:01:00 Sat 29 Sep 2018 UTC
03:01:00 Sun 30 Sep 2018 NZDT

14:02:00 Sat 29 Sep 2018 UTC
03:02:00 Sun 30 Sep 2018 NZDT

--------------------------------
12:57:00 Sat 06 Apr 2019 UTC
01:57:00 Sun 07 Apr 2019 NZDT

12:58:00 Sat 06 Apr 2019 UTC
01:58:00 Sun 07 Apr 2019 NZDT

12:59:00 Sat 06 Apr 2019 UTC
01:59:00 Sun 07 Apr 2019 NZDT

13:00:00 Sat 06 Apr 2019 UTC
01:00:00 Sun 07 Apr 2019 NZST

13:01:00 Sat 06 Apr 2019 UTC
01:01:00 Sun 07 Apr 2019 NZST

13:02:00 Sat 06 Apr 2019 UTC
01:02:00 Sun 07 Apr 2019 NZST

--------------------------------
13:57:00 Sat 28 Sep 2019 UTC
01:57:00 Sun 29 Sep 2019 NZST

13:58:00 Sat 28 Sep 2019 UTC
01:58:00 Sun 29 Sep 2019 NZST

13:59:00 Sat 28 Sep 2019 UTC
01:59:00 Sun 29 Sep 2019 NZST

14:00:00 Sat 28 Sep 2019 UTC
03:00:00 Sun 29 Sep 2019 NZDT

14:01:00 Sat 28 Sep 2019 UTC
03:01:00 Sun 29 Sep 2019 NZDT

14:02:00 Sat 28 Sep 2019 UTC
03:02:00 Sun 29 Sep 2019 NZDT

--------------------------------
```

stinkpot, 2018-11-11:
Thanks JChristensen,

Will try again

datacute, 2018-11-11:
Note that NZ changes from 3am to 2am (and back) not from 2am to 1am, but that means the rule is wrong, not the code's implementation of the rule.

JChristensen, 2018-11-11:
@datacute, thank you, I missed that. So the rules should be coded as:
```c++
// New Zealand Time Zone
TimeChangeRule myDST = {"NZDT", Last, Sun, Sep, 2, 780};    // UTC + 13 hours
TimeChangeRule mySTD = {"NZST", First, Sun, Apr, 3, 720};   // UTC + 12 hours
Timezone myTZ(myDST, mySTD);
```
http://www.legislation.govt.nz/regulation/public/2007/0185/latest/whole.html#DLM442250

JChristensen, 2018-11-11:
Test sketch and output above updated with the correct rules.

stinkpot, 2018-11-12:
Hi JChristensen,

I finally found the bug is in time.h make_time, that
static  const uint8_t monthDays[]={31,28,31,30,31,30,31,31,30,31,30,31}; 
is not constructed properly on UNO during running, maybe is my program is too large. Any way fixed it. Now all calculation is right now.

JChristensen, 2018-11-12:
Thanks for the update. Glad to hear you got things sorted out.

closed 2018-11-12 by JChristensen