JChristensen/Timezone #43: ESP32-board writeRules : undefined reference

remcoNL, 2018-12-29:
Hi Jack,

I'm using the Arduino IDE on an ESP32 board. It seemed to work fine, until I tried to write the rules to EEPROM. I checked your example 'WriteRules.ino' and it gives the same error.  Is it suppose to work on an ESP32..?

I get this error:

` undefined reference to `Timezone::writeRules(int)'`

JChristensen, 2018-12-29:
I'm not very familiar with the ESP boards. There was an earlier issue with ESP8266. At that time at least, EEPROM was not supported for the ESP boards (has that changed?), so [the EEPROM functions were disabled](https://github.com/JChristensen/Timezone/commit/510ae2f6b6fd64a2fc83ed5cee1c31c452daa7a1) in this library for architectures other than AVR.

remcoNL, 2018-12-29:
Jack your so fast  ! (-; 
EEPROM works fine in my experience on the ESP32,  I use [EEPROM.h](https://github.com/espressif/arduino-esp32/tree/master/libraries/EEPROM) for it.

JChristensen, 2018-12-29:
Been playing on the computer most of the day. Well it's been a while, maybe EEPROM wasn't supported initially but is now. It looks like that Espressif EEPROM library is fairly recent. Assuming it's compatible with the Arduino EEPROM library from an API standpoint, you could try taking the preprocessor directives out of the Timezone library. If you try that and it works, let me know and I can change the Timezone library.

remcoNL, 2018-12-30:
I tried different things but the EEPROM commands are different and in the end it didn't work. 

For me  I expect it's easier to try and store the values in a seperate routine...

per1234, 2018-12-30:
The ESP8266 and ESP32 EEPROM library API is compatible with the [Arduino AVR Boards EEPROM library](https://www.arduino.cc/en/Reference/EEPROM) but there is a difference (see https://arduino-esp8266.readthedocs.io/en/latest/libraries.html#eeprom, which also applies to ESP32). The ESP8266 and ESP32 don't have real EEPROM so flash memory is used to simulate EEPROM. In order to avoid unnecessary wear on the flash, the values written to "EEPROM" are stored in RAM until either `EEPROM.commit()` or `EEPROM.end()` is called. The user could call `EEPROM.commit()` in their sketch after calling `Timezone::writeRules()`, but they might not realize that's necessary, which could result in a lot of confusion. Instead, you could do at the end of `writeRules()`:

```c++
#if defined(ESP8266) || defined(ESP32)
    EEPROM.commit();
#endif  //defined(ESP8266) || defined(ESP32)
```

The other difference is that `EEPROM.begin()` must be called to set the size of the "EEPROM". I think it makes more sense to require that the user do that in their sketch.

remcoNL, 2018-12-30:
I added EEPROM.commit()  it in the sketch, but I'm not sure the API is really compatible:

`'eeprom_write_block' was not declared in this scope`

I tried to change it to EEPROM.put, but it didn't work either...

JChristensen, 2018-12-30:
Cannot support ESP hardware at this time. Sorry.

closed 2018-12-30 by JChristensen