JChristensen/Timezone #45: TimeChangeRules Brazil !!

Dancopy, 2019-03-15:
I would like to know what would be the example rule below for Brazil:

// Zone reference "usCT" US Central Time Zone (Chicago, Houston)
TimeChangeRule usCDT = {"CDT", Second, dowSunday, Mar, 2, -300};
TimeChangeRule usCST = {"CST", First, dowSunday, Nov, 2, -360};
Timezone usCT(usCDT, usCST);

Since, in Brazil, the Standard Time is -3 and the Daylight Saving Time is -2;

And how should these initials be:

TimeChangeRule usCDT = {"CDT", ...........
TimeChangeRule usCST = {"CST", ...........

Thanks

JChristensen, 2019-03-16:
Referring to https://en.wikipedia.org/wiki/Daylight_saving_time_in_Brazil
I would code:
```c++
// Brazil (Brasilia, Fortaleza)
TimeChangeRule brDST = {"BRST", First, Sun, Nov, 0, -120};    // Daylight time = UTC-2 hours
TimeChangeRule brSTD = {"BRT", Third, Sun, Feb, 0, -180};     // Standard time = UTC-3 hours
Timezone br(brDST, brSTD);
```

Dancopy, 2019-03-16:
Many thanks J Christensen

closed 2019-03-16 by JChristensen