JChristensen/Timezone #46: variable declared void

flyusa1, 2019-05-27:
Thank you for your work. I'm new to this and I am unable to verify the example sketches; the TimeZone, the TZ1 and TZ2 sketches all share a common compile error, "variable or field 'printDateTime' declared void". 

What am I doing wrong or what may I be missing? I have Timezine.h, the JC_Button.h, and the Streaming.h libraries installed. 

Again I'm new to this and still learning, thank you for your patience.

JChristensen, 2019-05-27:
Hello,
I don't seem to be able to reproduce the problem. What board are you using and what version of the IDE?
I'm confused by the names of the sketches (TimeZone, TZ1, TZ2). The example sketches that are part of the library are:
Change_TZ_1,  Change_TZ_2,  Clock,  HardwareRTC,  tzTest,  WorldClock,  WriteRules

flyusa1, 2019-05-27:
Hello, thank you. 

Pardon the confusion, I meant to type the TimeZone's HardwareRTC, Change_TZ_1 & Change_TZ_2. 

I resolved the HardwareRTC verification errors by relocating the TimeZone dot-h and dot-cpp files up one directory level from the src subdirectory. But I still have the same issue with the Change_TZ_1 & Change_TZ_2 sketches. 

I also relocated up one level, the the dot-h and dot-cpp files from the src subdirecotry of the JC_Button library.

I am using the Uno and IDE version 1.0.5-r2.

Thank you again. Still learning.

JChristensen, 2019-05-27:
1.0.5 is pretty old. The current version is 1.8.9. Why not use it?

flyusa1, 2019-05-27:
Yes, I will update; it is on an old and slow laptop.

JChristensen, 2019-05-27:
Windows machine?

flyusa1, 2019-05-27:
Yes, Windows 10.

JChristensen, 2019-05-27:
Well I don't imagine that there will be much of a difference in performance between the two versions but Win 10 on an old machine is probably less than ideal. Ever try Linux? I run it on some very old machines and it runs great and Arduino is very fast, faster than on Windows on a much newer machine.

flyusa1, 2019-05-27:
No, but thank you for the idea; w10 on an old lap top ages one quickly it is rather slow. the 10-install was an accident I was on 8.1 pro and it installed by itself before i found out and I couldn't roll back because the restoration files were incomplete - somehow. Never tried Linux, have heard lots about it. 

Are these errors below from the TZ1 verification due to library file structure or something else?

In file included from TimeZone_Example_Change_TZ_1.ino:28:
C:\Program Files\Arduino\libraries\Timezone/Timezone.h:41: error: 'time_t' does not name a type
C:\Program Files\Arduino\libraries\Timezone/Timezone.h:42: error: 'time_t' does not name a type
C:\Program Files\Arduino\libraries\Timezone/Timezone.h:43: error: 'time_t' does not name a type
C:\Program Files\Arduino\libraries\Timezone/Timezone.h:44: error: 'time_t' has not been declared
C:\Program Files\Arduino\libraries\Timezone/Timezone.h:45: error: 'time_t' has not been declared
C:\Program Files\Arduino\libraries\Timezone/Timezone.h:53: error: 'time_t' does not name a type
C:\Program Files\Arduino\libraries\Timezone/Timezone.h:56: error: 'time_t' does not name a type
C:\Program Files\Arduino\libraries\Timezone/Timezone.h:57: error: 'time_t' does not name a type
C:\Program Files\Arduino\libraries\Timezone/Timezone.h:58: error: 'time_t' does not name a type
C:\Program Files\Arduino\libraries\Timezone/Timezone.h:59: error: 'time_t' does not name a type
TimeZone_Example_Change_TZ_1:34: error: variable or field 'printDateTime' declared void
TimeZone_Example_Change_TZ_1:34: error: 'time_t' was not declared in this scope
TimeZone_Example_Change_TZ_1:35: error: 'time_t' does not name a type
TimeZone_Example_Change_TZ_1.ino: In function 'void setup()':
TimeZone_Example_Change_TZ_1:57: error: 'compileTime' was not declared in this scope
TimeZone_Example_Change_TZ_1:57: error: 'setTime' was not declared in this scope
TimeZone_Example_Change_TZ_1.ino: In function 'void loop()':
TimeZone_Example_Change_TZ_1:76: error: 'time_t' does not name a type
TimeZone_Example_Change_TZ_1:77: error: 'time_t' was not declared in this scope
TimeZone_Example_Change_TZ_1:77: error: expected `;' before 't'
TimeZone_Example_Change_TZ_1:78: error: 't' was not declared in this scope
TimeZone_Example_Change_TZ_1:78: error: 'tLast' was not declared in this scope
TimeZone_Example_Change_TZ_1:81: error: 'printDateTime' was not declared in this scope
TimeZone_Example_Change_TZ_1:84: error: 'class Timezone' has no member named 'toLocal'
TimeZone_Example_Change_TZ_1.ino: In function 'void changeTZ()':
TimeZone_Example_Change_TZ_1:101: error: 'eeprom_update_byte' was not declared in this scope
TimeZone_Example_Change_TZ_1.ino: At global scope:
TimeZone_Example_Change_TZ_1:109: error: variable or field 'printDateTime' declared void
TimeZone_Example_Change_TZ_1:109: error: 'time_t' was not declared in this scope
TimeZone_Example_Change_TZ_1:119: error: 'time_t' does not name a type

The HardwareRTC sketch works.

Thank you

JChristensen, 2019-05-27:
Is the Time library installed? Get it here: https://github.com/PaulStoffregen/Time

flyusa1, 2019-05-27:
Yes, I do have that. However I will reinstall from the above. Doing so now.

JChristensen, 2019-05-27:
Not sure what's going on, I suppose it could be library structure but 1.8.9 should work with or without the src subdirectory. Reinstalling the libraries sounds good. I'm signing off for tonight, good luck!

flyusa1, 2019-05-27:
Still the same. I will keep at it. I'll look at the later version too.

Thank you good night

closed 2019-05-27 by flyusa1