JChristensen/Timezone #47: BUG: Time time displayed for UTC+08:00 timezone (MYT) is 1 hour in future

alidrus, 2019-09-02:
I am trying to use this library to display time for my timezone which is UTC+8 but it appears the time obtained after conversion is 1 hour in the future.

**Expected result**:
Wednesday, 03/09/2019 04:46:40

**Actual result**:
Wednesday, 03/09/2019 05:46:40

The code I am running is this:
```c++
// vim: autoindent smartindent syntax=c expandtab tabstop=4 shiftwidth=4 softtabstop=4:

#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <String.h>
#include <NTPClient.h>
#include <Time.h>
#include <TimeLib.h>
#include <Timezone.h>

#define WIFI_SSID      "**********"        // WiFi Network SSID (name)
#define WIFI_PASSWORD  "**********"        // WiFi Network Password

#define NTP_SERVER     "asia.pool.ntp.org" // NTP Server
#define NTP_OFFSET     3600                // NTP Time Offset (60 minutes in seconds)
#define NTP_INTERVAL   60000               // NTP Interval (60 seconds in milliseconds)

// Instantiate UDP connection object
WiFiUDP ntpUDP;

// Instantiate NTP client object
NTPClient timeClient(ntpUDP, NTP_SERVER, NTP_OFFSET, NTP_INTERVAL);

const char * days[] = {"Sunday", "Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday"} ;

void setup(){
    Serial.begin(115200);

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

    while (WiFi.status() != WL_CONNECTED) {
        delay ( 500 );
        Serial.print ( "." );
    }

    timeClient.begin();
}

void loop() {
    if (WiFi.status() == WL_CONNECTED)
    {   
        timeClient.update();

        unsigned long epochTime =  timeClient.getEpochTime();

        time_t utc = epochTime;

        TimeChangeRule myMYT = {"MYT", Second, Sun, Mar, 2, 480};  // UTC + 8 hours

        Timezone malaysianTime(myMYT, myMYT);

        time_t local;

        local = malaysianTime.toLocal(utc);

        char dateTime[32];

        sprintf(dateTime,
                "%s, %02d/%02d/%04d %02d:%02d:%02d",
                days[weekday(local)],
                day(local),
                month(local),
                year(local),
                hour(local),
                minute(local),
                second(local));

        Serial.println(dateTime);
    }

    delay(1000);
}
```

JChristensen, 2019-09-02:
Could this be the issue, why is it necessary to adjust the NTP time?
```c++
#define NTP_OFFSET     3600                // NTP Time Offset (60 minutes in seconds)
```
This code gives the expected results:
```c++
#include <Timezone.h>   // https://github.com/JChristensen/Timezone

// Malaysia Time
TimeChangeRule myMYT = {"MYT", Second, Sun, Mar, 2, 480};  // UTC + 8 hours
Timezone myTZ(myMYT, myMYT);
TimeChangeRule *tcr;        // pointer to the time change rule, use to get TZ abbrev

void setup()
{
    Serial.begin(115200);
    setTime(20, 46, 40, 2, 9, 2019);
    time_t utc = now();
    time_t local = myTZ.toLocal(utc, &tcr);
    Serial.println();
    printDateTime(utc, "UTC");
    printDateTime(local, tcr -> abbrev);
}

void loop()
{
}

// format and print a time_t value, with a time zone appended.
void printDateTime(time_t t, const char *tz)
{
    char buf[32];
    char m[4];    // temporary storage for month string (DateStrings.cpp uses shared buffer)
    strcpy(m, monthShortStr(month(t)));
    sprintf(buf, "%.2d:%.2d:%.2d %s %.2d %s %d %s",
        hour(t), minute(t), second(t), dayShortStr(weekday(t)), day(t), m, year(t), tz);
    Serial.println(buf);
}
```
Output:
```
20:46:40 Mon 02 Sep 2019 UTC
04:46:40 Tue 03 Sep 2019 MYT
```
Also note, 03Sep is Tuesday on my calendar, not Wednesday.

alidrus, 2019-09-03:
You were right about NTP_OFFSET. 

My code was based off from [this instructable](https://www.instructables.com/id/Simplest-ESP8266-Local-Time-Internet-Clock-With-OL/) without very much modification other than stripping off the Wire.h and OLED stuff.

I have no idea why they set the 1 hour NTP offset. Even the examples in the NTPClient library also use the same value offset.

Anyway, I have since cleaned up my code (see below) and it worked perfectly. Sorry if this wasted your time. I guess I probably shouldn't have been working on this at 4:46 AM.

One more thing I would like to mention is when compiling I am getting this: "WARNING: library Timezone claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (esp8266) architecture(s)."

Your library seems to work flawlessly on esp8266 so perhaps the warning is unnecessary and you can include esp8266 into your list of architectures.

```
// vim: autoindent smartindent syntax=c expandtab tabstop=4 shiftwidth=4 softtabstop=4:

#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <String.h>
#include <NTPClient.h>
#include <Time.h>
#include <TimeLib.h>
#include <Timezone.h>

// Compiler defines
#define WIFI_SSID      "*************"     // WiFi Network SSID (name)
#define WIFI_PASSWORD  "**********"        // WiFi Network Password
#define NTP_SERVER     "asia.pool.ntp.org" // NTP Server
#define NTP_OFFSET     0                   // NTP Time Offset (in seconds)
#define NTP_INTERVAL   3600000             // NTP Interval (in milliseconds)

// Global variables
time_t utc, local;
char dateTime[34];

// Instantiate Class Objects (also global)
WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP, NTP_SERVER, NTP_OFFSET, NTP_INTERVAL);
TimeChangeRule myMYT = {"MYT", First, Sun, Mar, 2, 480};  // UTC + 8 hours
Timezone malaysianTime(myMYT, myMYT);

void setup(){
    Serial.begin(115200);

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

    while (WiFi.status() != WL_CONNECTED) {
        delay ( 500 );
        Serial.print ( "." );
    }

    timeClient.begin();
}

void loop() {
    if (WiFi.status() == WL_CONNECTED)
    {   
        timeClient.update();

        utc = (time_t) timeClient.getEpochTime();

        local = malaysianTime.toLocal(utc);

        sprintf(dateTime,
                "%s, %02d/%02d/%04d %02d:%02d:%02d %3s",
                dayStr(weekday(local)),
                day(local),
                month(local),
                year(local),
                hour(local),
                minute(local),
                second(local),
                myMYT.abbrev);

        Serial.println(dateTime);
    }

    delay(1000);
}
```

closed 2019-09-03 by alidrus