JChristensen/Timezone #51: Proper way to set local time using buttons.

foxsam21, 2020-03-09:
I have a clock that usually gets the time from wifi (ntp) but if the server is not reachable, the clock is portable, then it needs to be set manually. I did not try it yet but it looks like if I would use buttons to set the time it needs to set the utc time. What is the recommeded way to set the local time without the person setting it needing to know about utc time? Should I use `myTZ.toUTC(local);`?
Thanks

JChristensen, 2020-03-09:
I have at least one clock project which allows the user to set the time using local time, but then converts it to UTC using `toUTC()` for the purpose of setting the RTC.

As I noted in the README, there is some risk to this, but only if the user is setting the time during a transition period, i.e. daylight-to-standard time or vice versa. For me it was an acceptable risk ;-)

Cheers!

closed 2020-03-09 by JChristensen

foxsam21, 2020-03-09:
yes that's what I was wondering about. To me if something can go wrong it will. Maybe I'll check for that time and adjust for it.
Thanks.

JChristensen, 2020-03-09:
My application was just a simple clock sitting on a table in my house. Not a lot of impact if the time setting goes wrong. For something with a more serious purpose it may be worth some extra code to prevent an error.

foxsam21, 2020-03-09:
Its not a big deal at all except that it will fail the WAF test. Thats a big deal.