JChristensen/Timezone #57: Strange behavior compiled in windows

KalebClark, 2020-10-18:
First, thanks for this library loving it so far!

I developed most of this project in Linux, then I needed to do some other stuff in a Windows environment, so booted to that and needed to re flash the MCU, so I fired up Arduino in Windows, flashed it and everything that had a DST related was set to standard instead of DST. Drove me bonkers for an hour tracking it down. I verified that incoming NTP Unix time was correct, and verified that the 'time_t utc = now();' was returning correct Unix time. Only after that was it off. 

The only change that I had made, was compiling in Windows vs Linux. So I booted into Linux, compiled, flashed and the time was correct. 

Mostly an FYI. 

Thanks,
-Kaleb

JChristensen, 2020-10-18:
Hello Kaleb,

Glad that you like the library. That's an interesting issue but hard to comment on without more details. I've used the library both from Windows and Linux and it doesn't seem that where it's compiled should matter. What is important is the time reference that the MCU is using (sounds like NTP in your case), which should operate independently of the machine the code was compiled on.

I'll close this issue, but if you can supply a short example that I can use to replicate the problem, then we can reopen it.

Cheers ... Jack

closed 2020-10-18 by JChristensen

Bolukan, 2020-10-18:
It is common behaviour that after running linux (guess it uses utc + offset for local time or dstj). Windows fails to check and adjust time at boottime and as you discovered applies another mechanic.

JChristensen, 2020-10-18:
I'm aware that Linux sets the PC's RTC to UTC while Windows sets it to local time by default (but this can be changed.) In a dual-boot situation this can cause the PC's clock to be incorrect.
What I am not understanding is what that has to do with the microcontroller. Typically the MCU does not use a PC as its time source, but rather, its own RTC, NTP, or GPS, etc.
If I'm still missing the point, and if anyone can provide me with a short sketch that demonstrates the issue, I'd be glad to have a look.