JChristensen/Timezone #62: ESP8266 Support(?)

BCChad, 2022-03-14:
Hi,
I have found and thoroughly enjoyed your TIMEZONE, as well as the excellent documentation.  I have been struggling with a project clock involving the ESP8266; and trying to get the correct Daylight savings time (DST) conversion for the US.  Although it now compiles with your library and runs, I do note that I get a warning about "WARNING: library Timezone claims to run on avr architecture(s) and may be incompatible with your current board..."   
I get the correct DST conversion but performance is slow, so I'm thinking there may be a tweak needed on my part.
Can you comment on what in your library might be adjusted for ESP?

[And yes, I am aware of what the other fellow commented on using POSIX for this - I tried going down that path but was not getting the breakthrough (I may go back to that however).]

JChristensen, 2022-03-14:
Hi @BCChad,

Thanks for your note. I'm surprised that performance is an issue, since the library doesn't do anything that's computationally intensive. Can you narrow it down a little for me? Is performance slow when making a timezone conversion, i.e. calling `Timezone::toLocal()`? Have you tried any example sketches, are they slow too?

As for the warning, if you change [this line in the library.properties file](https://github.com/JChristensen/Timezone/blob/7cf1425b4ee598573e85b6fa124d4514f71c2586/library.properties#L9) to read
```c++
architectures=*
```
then the warning will go away. I may change this sometime in the future if I'm in making other changes.

JChristensen, 2022-03-14:
I'm not familiar with Tickers or ESP8266 generally.

I instrumented the Clock example sketch as follows, and running on an Arduino Uno (16MHz) the `toLocal()` call takes just over 2300 microseconds. It should be significantly faster than that on an ESP8266.

I'm wondering whether there are delays or other blocking code elsewhere that are causing the issues you have observed. Sorry I can't be of more help, good luck!

```c++
    uint32_t t1 = micros();
    time_t local = myTZ.toLocal(utc, &tcr);
    Serial.println(micros() - t1)
```

closed 2022-03-14 by JChristensen