JChristensen/Timezone #63: tcr->offset fails without calling toLocal prior to it

sezanzeb, 2022-04-03:
Can you please add a getOffset function that does the necessary steps to actually get the offset? This is a hidden dependency

```
    time_t local = myTZ.toLocal(utc, &tcr);
    Serial.print(tcr->offset);
```

works

```
    time_t local = myTZ.toLocal(utc, &tcr);
```

fails with a core dump / backtrace.

JChristensen, 2022-04-03:
`tcr` is a pointer to a `TimeChangeRule`.

Calling `toLocal()` returns a pointer to the `TimeChangeRule` that was used to do the time conversion. The idea is to let the caller know _which_ rule was used for the conversion.

In the second code snippet above, it would appear that `tcr` was never given a value, since `toLocal()` had not been called to do a conversion.

Dereferencing an empty pointer is sure to cause trouble!

sezanzeb, 2022-04-03:
> Dereferencing an empty pointer is sure to cause trouble!

I see. low-level programming is still a mystery to me

JChristensen, 2022-04-03:
Pointers are powerful but can be dangerous.
Why not just use the first code snippet? I don't see a downside.

sezanzeb, 2022-04-03:
yeah sure I'm using that one (https://github.com/sezanzeb/arduino-wxa50-alarm/blob/main/arduino-wxa50-alarm/time.cpp).

I just ran into the issue and it took some time until I figured out it was because I removed the call to `toLocal` previously. Thanks for pointing out that the pointer was empty.

closed 2022-04-03 by JChristensen