JChristensen/Timezone #65: Lack of ESP32 platform support

klosso, 2022-09-23:
I have worked with clock based on esp32 board with this very good lib, but there is lack of support for EEPROM for writing and reading. 
I have locally added support but, don't know haw to push this with review.

here is pure diff if You are interest :
diff --git a/library.properties b/library.properties
index 983999e..6e1c9c1 100644
--- a/library.properties
+++ b/library.properties
@@ -6,5 +6,5 @@ sentence=Arduino library to facilitate time zone conversions and automatic dayli
 paragraph=The primary aim of the Timezone library is to convert Universal Coordinated Time (UTC) to the correct local time, whether it is daylight saving time (a.k.a. summer time) or standard time.
 category=Timing
 url=https://github.com/JChristensen/Timezone
-architectures=avr
+architectures=avr,esp32
 depends=Time
diff --git a/src/Timezone.cpp b/src/Timezone.cpp
index fbf4fdf..8f4da0f 100644
--- a/src/Timezone.cpp
+++ b/src/Timezone.cpp
@@ -11,7 +11,9 @@
 #ifdef __AVR__
     #include <avr/eeprom.h>
 #endif
-
+#ifdef ESP_PLATFORM
+    #include <EEPROM.h>
+#endif
 /*----------------------------------------------------------------------*
  * Create a Timezone object from the given time change rules.           *
  *----------------------------------------------------------------------*/
@@ -31,7 +33,7 @@ Timezone::Timezone(TimeChangeRule stdTime)
         initTimeChanges();
 }
 
-#ifdef __AVR__
+#if defined(__AVR__) || defined(ESP_PLATFORM)
 /*----------------------------------------------------------------------*
  * Create a Timezone object from time change rules stored in EEPROM     *
  * at the given address.                                                *
@@ -216,6 +218,32 @@ void Timezone::setRules(TimeChangeRule dstStart, TimeChangeRule stdStart)
     initTimeChanges();  // force calcTimeChanges() at next conversion call
 }
 
+#ifdef ESP_PLATFORM
+/*----------------------------------------------------------------------*
+ * Read the daylight and standard time rules from EEPROM at             *
+ * the given address.                                                   *
+ *----------------------------------------------------------------------*/
+void Timezone::readRules(int address)
+{
+    EEPROM.readBytes((int) &m_dst, (char *) address, sizeof(m_dst));
+    address += sizeof(m_dst);
+    EEPROM.readBytes((int) &m_std, (char *) address, sizeof(m_std));
+    initTimeChanges();  // force calcTimeChanges() at next conversion call
+}
+
+/*----------------------------------------------------------------------*
+ * Write the daylight and standard time rules to EEPROM at              *
+ * the given address.                                                   *
+ *----------------------------------------------------------------------*/
+void Timezone::writeRules(int address)
+{
+    EEPROM.writeBytes((int) &m_dst, (char *) address, sizeof(m_dst));
+    address += sizeof(m_dst);
+    EEPROM.writeBytes((int) &m_std, (char *) address, sizeof(m_std));
+    EEPROM.commit();
+}
+
+#endif
 #ifdef __AVR__
 /*----------------------------------------------------------------------*
  * Read the daylight and standard time rules from EEPROM at             *

FallenRecruit, 2022-10-02:
any chance I could get a link to the files that you have changed or a better paste of the code changes / additions

scarolan, 2023-02-10:
Me too! Or maybe we can help you get this into a pull request, @klosso