JChristensen/Timezone #67: Error in hour character  display

codeput, 2024-04-13:
Hi, I managed to program my ESP correctly, but the time string appears like this "08:18:33.045 -> ============    17 55  Saturday ---- 13 4 2024" in practice the hours are not visible and I think that for this reason everything doesn't work.

JChristensen, 2024-04-13:
I cannot respond if I cannot duplicate the issue. If you would provide a [SSCCE](http://www.sscce.org/), that would help greatly.

codeput, 2024-04-13:
Hi, thanks for the quick reply, I solved it by adding an empty character on the string "SERIALPORT.println(String(" ") + hour(local) + String(" ") + minute(local) + String(" ") + second (local) + String(" ") + String(weekdays[weekday(local)-1]) + String(" ---- ") + day(local) + String(" ") + month(local) + String (" ") + year(local));"
Thanks again.

closed 2024-04-13 by codeput

JChristensen, 2024-04-13:
Hi, thanks for the update. Glad you found your issue.